MikroElektronika/mikrosdk_v2 #3: [FEATURE] AVR / Arduino support

techdragon, 2022-07-25:
***Is your feature request related to a problem? Please describe.**
The MikroBus shields for Arduino are some of the best low cost ways to get started with the entire range of click boards. Not having support for them in the current version of mikroSDK adds significant friction.   [...]

**Describe the solution you'd like**
The mikroSDK to support AVR microcontrollers and add support for projects using the Arduino MikroBus shields. 

**Describe alternatives you've considered**
I've looked at porting it myself, but I'm not familiar enough with the codebase for the size of the task.

**Additional context**
N.A.

StrahinjaJacimovic, 2022-07-25:
Hello @techdragon,
We will take this into consideration.

I will leave the issue open and contact you once an adequate solution has been found.

We will consider adding AVR to mikroSDK next.

I cannot give you a concrete timeframe as to when you can expect the feature, but please be patient.

If you have any other suggestions, feel free to post them.

Thank you for using mikroSDK.
Regards,
Strahinja Jacimovic.

StrahinjaJacimovic, 2022-12-01:
Hello @techdragon 

We have added support for AVR MCUs in the mikroSDK.

Please have a look at the implementation changes for the latest release.

[RELEASE 2.7.0](https://github.com/MikroElektronika/mikrosdk_v2/releases/tag/mikroSDK-2.7.0)

If you have a specific request for a board or board/shield combination to be added, please create another issue with your request here and we will look into it.

I am now closing this issue.

Regards,
Strahinja Jacimovic.

closed 2022-12-01 by StrahinjaJacimovic