MikroElektronika/mikrosdk_v2 #4: [BUG]VTFT active instance

StrahinjaJacimovic, 2022-11-28:
**Describe the bug**

- Accessing any component on `PIC18F57Q43` results in incorrect instance handler
- Incorrect instance handler in turn results in the component not being used appropriately

**To Reproduce**

- Create a setup for:
  - Compiler: `mikroC AI for PIC 1.1.1`
  - Board: `EasyPIC PRO v8`
  - MCU: `MCU CARD 11 for PIC PIC18F57Q43`
  - Display: Any display
- Create a new `DESIGNER` project with this setup
- Add a button to the project, build it and program it
- Click on the button on the display

**Expected behavior**

+ Button visibly changes color to indicate it has been clicked

**Information:**
 - OS: Windows 10
 - SDK `2.6.0`
 - HW:
   - `EasyPIC PRO v8`
   - `MCU CARD 11 for PIC PIC18F57Q43`

**Additional context**

- Issue seems to be related to `PIC18F57Q43` specifically

StrahinjaJacimovic, 2022-12-01:
Implemented workaround for this.

[Following line](https://github.com/MikroElektronika/mikrosdk_v2/blob/master/api/vtft/lib_vtft/src/vtft_touch.c#L56) is the workaround.

MarkoZiv, 2022-12-28:
Solved in v2.7.1

closed 2022-12-28 by MarkoZiv