MikroElektronika/mikrosdk_v2 #7: STM32 CAN  Function support

skylin008, 2023-11-11:
***Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

skylin008, 2023-11-16:
Thanks @StrahinjaJacimovic

StrahinjaJacimovic, 2024-01-30:
Hello @skylin008 

CAN has been added for all architectures. Best regards and have fun using it.

Closing the issue now.

closed 2024-01-30 by StrahinjaJacimovic