MikroElektronika/mikrosdk_v2 #9: microC AI for ARM does not run with Clicker4 STM32F4 board

js1tr3, 2024-04-04:
**Describe the bug**
blink example from website does not run with MicroC AI for ARM

#include "board.h"
#include "drv_digital_out.h"

static digital_out_t led;

/// @brief Application init function.
void application_init ()
{
    digital_out_init(&led, LED_1);
}

/// @brief Application task.
void application_task ()
{
    digital_out_toggle(&led);
    Delay_ms(1000);
}

/// @brief Application main function.
void main (void)
{

    application_init();

    while (1)
    {
        application_task();
    }
}

**To Reproduce**
Compile and download blink example
Run with microC AI arm (code does not execute no blink)
Run with GCC for arm code runs, LED blinks as expected

**Information (please complete the following information):**
Windows]
SDK version 2.11
 - HW related to issue (board, MCU, etc.) STM32F4 Clicker4

**Additional context**
New update

js1tr3, 2024-04-04:
Correction, Flashing works and the code runs, but cant debug it.

StrahinjaJacimovic, 2024-04-05:
Hello @js1tr3 

Thank you for the detailed bug report.
We have looked into it and this seems to be a CODEGRIP relate issue, not mikroSDK.
We are working on finding a prompt and appropriate solution.

I will leave the issue open and update you as soon as we have a fix available.

Kind regards,
Strahinja Jacimovic.

MarkoZiv, 2024-04-05:
Hello @js1tr3, 

We have a solution for you, 
- download https://tinyurl.com/2c8kcuvn (it is updated version of codegrip-dll-core.dll)
- copy it to the folder you have installed CODEGRIP, inside \CODEGRIP Suite\ subfolder
and let us know if the issue is solved.

Regards
Marko

js1tr3, 2024-04-05:
Yup that Fixed it, Thankyou!